moeiscool/Shinobi #159: Syntax unexpected token >

megashot, 2017-05-23:
Hello,

Today i installed it using the INSTALL/ubuntu.sh following the ubuntu easy way (http://shinobi.video/docs/start) and whenever there is nodemailer.sendMail(d.mailOptions, (error, info) => 
Syntax unexpected token > is produced.

moeiscool, 2017-05-23:
Please ensure you are on node.js version 7.2.0 or above. You can confirm this by doing the following
```
node -v
```

if you are in need to update run this 
```
sudo npm cache clean -f
sudo npm install -g n
sudo n stable
```

once done that navigate to your shinobi directory and run
```
npm install
```

megashot, 2017-05-24:
Hi,

When i do node -v i get v0.10.29, i did the update procedure

moeiscool, 2017-05-24:
please try rebooting your system if it still appears the same.

closed 2017-05-24 by moeiscool