moeiscool/Shinobi #168: Install Guide Incorrect

clmarshall, 2017-05-28:
Kitematic directions are inaccurate.  Kitematic is now part of Docker Toolbox.  After installing pre-reqs on Windows 10, Docker and DockerToolbox and  following directions the Container Log reads "Waiting for MYSQL" and does not progress past that point.

I've manually installed MySQL including shinobi account settings found in the settings of the shinobi container with no effect.

moeiscool, 2017-05-28:
I just tried it and it worked for me. Did you make sure to select the `kitematic` tag? `latest` and `dev` don't work with kitematic because docker didn't update kitematic to work with the newer docker standards.

closed 2017-05-28 by moeiscool

moeiscool, 2017-05-28:
Here is a video of me doing it with the new tag https://youtu.be/k8sWHRWiaCM