moeiscool/Shinobi #182: Unable to login to Shinobi for the first time.

svaguilarv, 2017-06-05:
Hi,

This is the first install I've done with Shinobi which I installed on a Ubuntu Server 16.04 machine.

Everything seems to be running  well except when I try to login for the first time 
[image]

but when I try to login with the default login:
Username : ccio@m03.ca
Password : password

Nothing Happens.

But when I try the super User it does login into the SuperUser mode but when i try to add a new user it takes me back to the login page without any error as below.

[image]

[image]

[image]

When I try to login with the new user it returns me to the login page

[image]

[image]

Any Help with this Issue?

moeiscool, 2017-06-05:
few questions:
1. what method did you use to install?
2. can you get into your sql via command line or some other tool?
2a. if you can get in please check if you have the tables set up. 
 - `framework.sql` - the table structure for the program, no rows are added.
 -  `user.sql` - the user that the conf.json uses to login to the SQL server.
 - `default_data.sql` - **OPTIONAL** this has a demo camera and the user you are trying to login with `ccio@m03.ca`

The issue with your superuser login is that it is unable to get data from the server. Which leads me to believe your SQL tables aren't set up.

The easiest way to set up sql files is to go to shinobi directory then `./sql`. Login to mysql command line like this.
```
mysql -u root -p
```

Once there you can run 
```
source framework.sql
source user.sql
source default_data.sql
```

svaguilarv, 2017-06-05:
1. I installed it using the easy way

2.- cannot get into the login using ccio@m03.ca 

I can get into SQL
[image]

[image]

After 
pm2 flush
pm2 restart camera.js
pm2 logs

I get this now:

[image]

:'(

moeiscool, 2017-06-05:
go to shinobi directory.

`Process camera.js not found` because its not already posted to the list you got in the image there.
`pm2 start camera.js` to start it.

you must be inside the main shinobi directory for `pm2 [SOMETHING] camera.js` to work or you will get a similar error to the one you have shown in the image above.

moeiscool, 2017-06-05:
once shinobi is started please post your logs as well. you can get the logs by doing `pm2 logs`

svaguilarv, 2017-06-05:
This is what i get after rebooting the machine:

I also tried what you just posted:

[image]

moeiscool, 2017-06-05:
You are not in the shinobi directory, that's why it says `Script not found`. You are in your home directory. Please join me on discord, i can help you better there. http://discordapp.com/invite/mdhmvuH/ As this is general support, i don't want to keep blasting everyone's email boxes. Thanks :)

closed 2017-06-05 by moeiscool