moeiscool/Shinobi #193: Form base uri using initial http request for shinobi

mafischer, 2017-06-15:
base uri for requests initiated in browser should be formed using the protocol, domain, and port of initial page load. See error below.

Mixed Content: The page at 'https://cctv.example.com/' was loaded over HTTPS, but requested an insecure script 'http://localhost:8080/libs/js/socket.io.js'. This request has been blocked; the content must be served over HTTPS.

moeiscool, 2017-06-15:
how did you get this error? what did you do?

mafischer, 2017-06-15:
running behind a reverse proxy that terminates the ssl before forwarding on to shinobi on port 8080

moeiscool, 2017-06-15:
where are you getting this error?
does it happen without the proxy?
and does it happen on cloudflare?

mafischer, 2017-06-15:
This does not occur outside of using a reverse proxy. See proxy config below:
server {
        listen 80;
        listen [::]:80;
        server_name cctv.example.com;
        return 301 https://$host$request_uri;
}
server {
        listen 443;
        listen [::]:443;

        server_name cctv.fischerapps.com;

        location / {
                proxy_pass http://localhost:8080/;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
        }
}

mafischer, 2017-06-15:
This is occurring when you use the power viewer.

mafischer, 2017-06-15:
These are the the requests that are failling
http://localhost:8080/libs/js/socket.io.js
http://localhost:8080/libs/js/hls.min.js
http://localhost:8080/libs/js/jquery.min.js

speedst3r, 2017-06-19:
Also occurs when selecting regions for motion detection.

moeiscool, 2017-06-20:
the newest dev has a change in how the files are served. Cozza, on Discord, was able to get it working correctly through his proxy after i made the changes. You can see the change in this commit on `Line 2371` (green side) https://github.com/moeiscool/Shinobi/commit/6f00d0b52d1a7f9d65aabf3b6447972e1c8f75e4

yukihiratype2, 2017-07-07:
This problem still exists. I use apache2 as a reverse proxy like this:
`  
  ProxyPass /watch http://localhost:8080
  ProxyPassReverse /watch  http://localhost:8080
`
And shinobi can't working properly.
 Another server behind apache2 is working well after I tweak it's base URI.

moeiscool, 2017-07-07:
just a note i have added native SSL capabilities. If that is why you guys are using proxy please try it out.

speedst3r, 2017-07-08:
Thanks moe. Whilst it is great that Shinobi can now serve HTTPS directly there are other reasons why using a reverse proxy to publish it is desirable:
- network isolation (i.e. only access is via the reverse proxy)
- using the proxy as a "web firewall" (run request filters/connection limits to protect the application from exploits or DoS)
- offloading SSL to a more appropriate device (e.g. has hardware crypto or more powerful CPU)

If we could keep this issue active that would be appreciated 😃

Cozza38, 2017-07-09:
Remove ` <%=data.url%> ` on lines 11, 12 & 24 in ` /web/pages/embed.ejs ` at least partially resolves this issue.

Cozza38, 2017-07-09:
Replaced every instance of ` %=data.url%> ` in  ` embed.ejs ` and power viewer is now functioning correctly behind an SSL offload Nginx reverse proxy.

Probably breaks some other functionality though.

https://gist.github.com/Cozza38/665ca6e6756f0bd0a14b5c85dba3b56c

moeiscool, 2017-07-09:
#226

moeiscool, 2017-07-09:
since michael was running shinobi in a similar way as @Cozza38 and it now works for Cozza I am gonna close this issue. Thanks Cozza for finding the fix!

closed 2017-07-09 by moeiscool