moeiscool/Shinobi #203: Can't record a monitor

Pes8, 2017-06-22:
Hi, I have 2 monitor correctly configurated. When I press "record" to save the video in the filesystem a warning appear and the monitor is not really recording. (no files are saved in the disk)

this is the warning:

-loglevel warning -reconnect 1 -r 24 -f mjpeg -i http://XXX:YYY@ZZZ/mjpg/video.mjpg -vcodec libx264 -r 1 -s 640x480 -f segment -segment_atclocktime 1 -reset_timestamps 1 -strftime 1 -segment_list pipe:2 -segment_time 900 /usr/share/Shinobi/videos/2Df5hBE/YoHdmdKQBk/%Y-%m-%dT%H-%M-%S.mp4 -c:v mjpeg -f image2pipe -q:v 4 -r 24 -s 704x576 pipe:1 -f singlejpeg -vf fps=2  -s 320x240 pipe:0

No pixel format specified, yuvj420p for H.264 encoding chosen.
Use -pix_fmt yuv420p for compatibility with outdated media players.

moeiscool, 2017-06-22:
Are there more logs? This first bit is just your ffmpeg command and it looks fine. Are there any other logs? If it keeps exiting try building ffmpeg from source or getting from a repo.

Usually this happens because you are missing a library or the version of ffmpeg you have is damaged or missing options. You can check this by running `ffmpeg` in terminal. Paste the output here.

Also what OS and how did you install it?

Pes8, 2017-06-22:
OS:

`Ubuntu 16.04.2 LTS (GNU/Linux 4.4.0-81-generic x86_64)`

ffmpeg:

```
ffmpeg version 3.3.2-1~16.04.york1 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 20160609
  configuration: --prefix=/usr --extra-version='1~16.04.york1' --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 58.100 / 55. 58.100
  libavcodec     57. 89.100 / 57. 89.100
  libavformat    57. 71.100 / 57. 71.100
  libavdevice    57.  6.100 / 57.  6.100
  libavfilter     6. 82.100 /  6. 82.100
  libavresample   3.  5.  0 /  3.  5.  0
  libswscale      4.  6.100 /  4.  6.100
  libswresample   2.  7.100 /  2.  7.100
  libpostproc    54.  5.100 / 54.  5.100
```

Before I used a static build of ffmpeg but then I installed it from the repo ...
I can't find any other log. Only this "No pixel format specified, yuvj420p for H.264 encoding chosen.
Use -pix_fmt yuv420p for compatibility with outdated media players." but seem ore a warning than an error.

Are there some Shinobi specific logs to check ?

moeiscool, 2017-06-22:
hm. can you please post your settings. Just hit export on the monitor settings window and upload the JSON it makes.

you can save the logs for the monitor by enabling it in the green `Logging` section at the bottom of the monitor settings.

for logs about the node apps you can run `pm2 logs`

The warning is a flag for either record or input, i forgot. You can put the flag in the `Custom` section in either recording or input.

Pes8, 2017-06-22:
Thank you, I'll be on that machine next Tuesday and I'll post more informations

Pes8, 2017-06-27:
**Monitor Settings:**
```{
	"mode": "start",
	"mid": "5QFWfb6M62",
	"name": "VarcoIngresso",
	"type": "mjpeg",
	"protocol": "http",
	"host": "cam1in.XXX.loc",
	"port": "80",
	"path": "/cgi-bin/cmd/system?GET_STREAM",
	"ext": "mp4",
	"fps": "1",
	"width": "640",
	"height": "480",
	"details": "{\"fatal_max\":\"\",\"notes\":\"\",\"rtsp_transport\":\"tcp\",\"muser\":\"view\",\"mpass\":\"1234\",\"port_force\":null,\"sfps\":\"24\",\"aduration\":\"\",\"stream_type\":\"b64\",\"stream_mjpeg_clients\":\"\",\"stream_vcodec\":\"libx264\",\"stream_acodec\":\"no\",\"hls_time\":\"2\",\"preset_stream\":\"ultrafast\",\"hls_list_size\":\"3\",\"signal_check\":\"10\",\"signal_check_log\":\"0\",\"stream_quality\":\"8\",\"stream_fps\":\"24\",\"stream_scale_x\":\"720\",\"stream_scale_y\":\"576\",\"svf\":\"\",\"stream_server\":null,\"stream_server_url\":\"\",\"stream_server_fps\":\"\",\"stream_server_scale_x\":\"\",\"stream_server_scale_y\":\"\",\"stream_server_crf\":\"\",\"stream_server_preset\":\"\",\"stream_server_vbr\":\"\",\"stream_server_vf\":\"\",\"snap\":null,\"snap_fps\":\"\",\"snap_scale_x\":\"\",\"snap_scale_y\":\"\",\"snap_vf\":\"\",\"vcodec\":\"libx264\",\"crf\":\"1\",\"preset_record\":\"\",\"acodec\":\"none\",\"dqf\":\"0\",\"cutoff\":\"15\",\"vf\":\"\",\"timestamp\":\"0\",\"timestamp_font\":\"\",\"timestamp_font_size\":\"10\",\"timestamp_color\":\"white\",\"timestamp_box_color\":\"0x00000000@1\",\"timestamp_x\":\"(w-tw)/2\",\"timestamp_y\":\"0\",\"cust_input\":\"\",\"cust_snap\":\"\",\"cust_detect\":\"\",\"cust_stream\":\"\",\"cust_stream_server\":\"\",\"cust_record\":\"\",\"custom_output\":\"\",\"detector\":\"0\",\"detector_webhook\":null,\"detector_webhook_url\":\"\",\"detector_command_enable\":null,\"detector_command\":\"\",\"detector_command_timeout\":\"\",\"detector_lock_timeout\":\"\",\"detector_record_method\":null,\"detector_trigger\":null,\"detector_timeout\":\"\",\"watchdog_reset\":null,\"detector_delete_motionless_videos\":null,\"detector_save\":null,\"detector_mail\":\"0\",\"detector_mail_timeout\":\"\",\"detector_send_frames\":null,\"detector_fps\":\"\",\"detector_scale_x\":\"\",\"detector_scale_y\":\"\",\"detector_frame\":\"1\",\"detector_sensitivity\":\"\",\"cords\":\"\",\"detector_face\":null,\"detector_fullbody\":null,\"detector_car\":null,\"detector_notrigger\":null,\"detector_notrigger_mail\":null,\"detector_notrigger_timeout\":\"\",\"control\":\"0\",\"control_base_url\":\"\",\"control_stop\":\"0\",\"control_url_stop_timeout\":\"\",\"control_url_center\":\"\",\"control_url_left\":\"\",\"control_url_left_stop\":\"\",\"control_url_right\":\"\",\"control_url_right_stop\":\"\",\"control_url_up\":\"\",\"control_url_up_stop\":\"\",\"control_url_down\":\"\",\"control_url_down_stop\":\"\",\"control_url_enable_nv\":\"\",\"control_url_disable_nv\":\"\",\"control_url_zoom_out\":\"\",\"control_url_zoom_out_stop\":\"\",\"control_url_zoom_in\":\"\",\"control_url_zoom_in_stop\":\"\",\"groups\":\"[]\",\"loglevel\":\"warning\",\"sqllog\":\"0\"}",
	"shto": "[]",
	"shfr": "[]"
}
```

**pm2 logs:**

```0|camera   | uncaughtException { Error: read ECONNRESET
0|camera   |     at exports._errnoException (util.js:1050:11)
0|camera   |     at Pipe.onread (net.js:582:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
0|camera   | uncaughtException { Error: read ECONNRESET
0|camera   |     at exports._errnoException (util.js:1050:11)
0|camera   |     at Pipe.onread (net.js:582:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
0|camera   | uncaughtException { Error: read ECONNRESET
0|camera   |     at exports._errnoException (util.js:1050:11)
0|camera   |     at Pipe.onread (net.js:582:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
0|camera   | uncaughtException { Error: read ECONNRESET
0|camera   |     at exports._errnoException (util.js:1050:11)
0|camera   |     at Pipe.onread (net.js:582:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
0|camera   | uncaughtException { Error: read ECONNRESET
0|camera   |     at exports._errnoException (util.js:1050:11)
0|camera   |     at Pipe.onread (net.js:582:26) code: 'ECONNRESET', errno: 'ECONNRESET', syscall: 'read' }
```

**Monitor Log (level debug) on DB:**
```
{"type":"FFMPEG Process Started","msg":{"cmd":"-loglevel debug -reconnect 1 -r 24 -f mjpeg -i http://view:1234@cam1in.XXX.loc/cgi-bin/cmd/system?GET_STREAM -c:v mjpeg -f image2pipe -q:v 8 -r 24 -s 720x576 pipe:1"}}
{"type":"FFMPEG STDERR","msg":"ffmpeg version 3.3.2-1~16.04.york1"}
{"type":"FFMPEG STDERR","msg":" Copyright (c) 2000-2017 the FFmpeg developers"}
{"type":"FFMPEG STDERR","msg":"\n"}
{"type":"FFMPEG STDERR","msg":"  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 20160609\n"}
{"type":"FFMPEG STDERR","msg":"  configuration: --prefix=/usr --extra-version='1~16.04.york1' --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-li [... cut ...]
{"type":"FFMPEG STDERR","msg":"    Last message repeated 1 times\n[http @ 0x55f65a3f6080] HTTP error 401 Unauthorized\nhttp://view:1234@cam1in.XXX.loc/cgi-bin/cmd/system?GET_STREAM: Server returned 401 Unauthorized (authorization failed)\n"}
{"type":"FFMPEG Unexpected Exit","msg":{"msg":"Process Crashed for Monitor : 5QFWfb6M62","cmd":"-loglevel debug -reconnect 1 -r 24 -f mjpeg -i http://view:1234@cam1in.XXX.loc/cgi-bin/cmd/system?GET_STREAM -c:v mjpeg -f image2pipe -q:v 8 -r 24 -s 720x576 pipe:1"}}
```

Seem a problem of authentication, but I set User&Password correctly. (If I open in the broswer the monitor link I can login and view the stream)

moeiscool, 2017-06-27:
you aren't able to record because `copy` is an invalid selection for MJPEG streams. Based on your selections the following notice appears in the recording section.

```
It is recommended that you set Record File Type to MP4 and Video Codec to libx264 because your Input Type is set to MJPEG.
```

As this is not a bug with the software I will be closing this thread. If you need further assistance please find me over on the community chat.

closed 2017-06-27 by moeiscool