moeiscool/Shinobi #204: SQL errors in docker image

AdySan, 2017-06-23:
Hello, I'm running this as a docker image on Ubuntu 16.04 Server. I got audio to work with HLS, everything seems fine, until after a few hours, streaming stops and I see these errors in the docker log

```
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T16:59:53.214Z' } 
1|camera   | 2017-06-21T17:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T17:59:53.214Z' } 
1|camera   | 2017-06-21T17:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T17:59:53.214Z' } 
1|camera   | 2017-06-21T18:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T18:59:53.215Z' } 
1|camera   | 2017-06-21T18:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T18:59:53.215Z' } 
1|camera   | 2017-06-21T19:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T19:59:53.218Z' } 
1|camera   | 2017-06-21T19:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T19:59:53.219Z' } 
1|camera   | 2017-06-21T20:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T20:59:53.219Z' } 
1|camera   | 2017-06-21T20:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T20:59:53.219Z' } 
1|camera   | 2017-06-21T21:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T21:59:53.220Z' } 
1|camera   | 2017-06-21T21:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T21:59:53.220Z' } 
1|camera   | 2017-06-21T22:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T22:59:53.222Z' } 
1|camera   | 2017-06-21T22:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T22:59:53.223Z' } 
1|camera   | 2017-06-21T23:41:16+00:00 DB Lost.. Retrying..  
1|camera   | 2017-06-21T23:41:16+00:00 { Error: Connection lost: The server closed the connection.
1|camera   |     at Protocol.end (/opt/shinobi/node_modules/mysql/lib/protocol/Protocol.js:109:13)
1|camera   |     at Socket.<anonymous> (/opt/shinobi/node_modules/mysql/lib/Connection.js:109:28)
1|camera   |     at emitNone (events.js:110:20)
1|camera   |     at Socket.emit (events.js:207:7)
1|camera   |     at endReadableNT (_stream_readable.js:1045:12)
1|camera   |     at _combinedTickCallback (internal/process/next_tick.js:102:11)
1|camera   |     at process._tickDomainCallback (internal/process/next_tick.js:198:9) fatal: true, code: 'PROTOCOL_CONNECTION_LOST' }  
1|camera   | 2017-06-21T23:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T23:59:53.223Z' } 
1|camera   | 2017-06-21T23:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-21T23:59:53.224Z' } 
1|camera   | 2017-06-22T00:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T00:59:53.225Z' } 
1|camera   | 2017-06-22T00:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T00:59:53.226Z' } 
1|camera   | 2017-06-22T01:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T01:59:53.224Z' } 
1|camera   | 2017-06-22T01:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T01:59:53.225Z' } 
1|camera   | 2017-06-22T02:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T02:59:53.225Z' } 
1|camera   | 2017-06-22T02:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T02:59:53.226Z' } 
1|camera   | 2017-06-22T03:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T03:59:53.225Z' } 
1|camera   | 2017-06-22T03:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T03:59:53.226Z' } 
1|camera   | 2017-06-22T04:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T04:59:53.227Z' } 
1|camera   | 2017-06-22T04:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T04:59:53.228Z' } 
1|camera   | 2017-06-22T05:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T05:59:53.228Z' } 
1|camera   | 2017-06-22T05:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T05:59:53.228Z' } 
1|camera   | 2017-06-22T06:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T06:59:53.229Z' } 
1|camera   | 2017-06-22T06:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T06:59:53.229Z' } 
1|camera   | 2017-06-22T07:41:17+00:00 DB Lost.. Retrying..  
1|camera   | 2017-06-22T07:41:17+00:00 { Error: Connection lost: The server closed the connection.
1|camera   |     at Protocol.end (/opt/shinobi/node_modules/mysql/lib/protocol/Protocol.js:109:13)
1|camera   |     at Socket.<anonymous> (/opt/shinobi/node_modules/mysql/lib/Connection.js:109:28)
1|camera   |     at emitNone (events.js:110:20)
1|camera   |     at Socket.emit (events.js:207:7)
1|camera   |     at endReadableNT (_stream_readable.js:1045:12)
1|camera   |     at _combinedTickCallback (internal/process/next_tick.js:102:11)
1|camera   |     at process._tickDomainCallback (internal/process/next_tick.js:198:9) fatal: true, code: 'PROTOCOL_CONNECTION_LOST' }  
1|camera   | 2017-06-22T07:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T07:59:53.233Z' } 
1|camera   | 2017-06-22T07:59:53+00:00 CRON :  { f: 'deleteEvents',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T07:59:53.234Z' } 
1|camera   | 2017-06-22T08:59:53+00:00 CRON :  { f: 'deleteLogs',
1|camera   |   msg: '0 SQL rows older than 10 days deleted',
1|camera   |   ke: '2Df5hBE',
1|camera   |   time: '2017-06-22T08:59:53.231Z' } 
```

Restarting the docker image helped, but any clue what might be going on here. Let me know if there is any more information needed from my side.

moeiscool, 2017-06-23:
to my knowledge when this happens its usually the following : 
- using a remote SQL server and losing connection frequently
- drive already in heavy use
- drive dying
- sql is dying for some reason possibly getting too many requests from something

i don't know of what else might causes this.

what kind of specs are you running?

AdySan, 2017-06-23:
@moeiscool This docker container is running on an Intel NUC7i3BNK, with 16GB RAM and a 120GB M2SSD.  One reason could be is that there is another container running Plex on the same computer, that can have high disk activity when some video is being streamed. As for the other possibilities, I doubt any other process is connecting to the SQL server. AFAIK, everything used by this docker image should be contained within the image. Let me know if there are any options I can turn off to reduce SQL requests. All I really care about is real time video streaming, anything else like motion detection, etc. can be turned off if it would help in any way.

moeiscool, 2017-06-23:
yah i'm going to say it's Plex. I used to run it on my desktop and I've seen how much it eats resources.

well its not going to save anything to SQL if its just streaming. Motion doesn't do much if the plugin isn't connected. The kitematic tag is contained in one image.

Since this is not actually a shinobi bug im going to close this thread. If you want you can join me over on the discord community chat. https://discordapp.com/invite/mdhmvuH

closed 2017-06-23 by moeiscool