moeiscool/Shinobi #209: Power Video Viewer: Date Range dialog: Dropdowns for hour, minute and AM/PM not opening

DL44227, 2017-06-28:
Hello,

In the Power Video Viewer choosing a range with the Date Range dialog, it is not possible to select hours, minutes and AM/PM from dropdown elements (hourselect, minuteselect, ampmselect), because they don't open in Firefox. In Chromium they worked as expected.

Environment: Firefox 53.0 (64 Bit), Linux Mint

moeiscool, 2017-06-28:
could you please post your console log output, it should have some stuff in red.

DL44227, 2017-06-28:
It says: "Loading of mixed active content "http://my.domain:8080/libs/js/jquery.min.js" was blocked. My server is reachable via https://my.domain/... (apache reverse proxy). So maybe there are absolute http references in the code?

moeiscool, 2017-06-28:
```
<script src="/libs/js/jquery.min.js"></script>
```

this is how i define jquery.

Can you post everything? You should be able to click a little arrow next to the error for it to show more details. please do that to all the errors and post.

DL44227, 2017-06-28:
Please find a screenshot below
[image]

andersonls, 2017-06-29:
Same issue here, using Firefox 54.0 (64-bit) with Linux Mint. The Shinobi, however, is running on docker with the image shinobicctv/shinobi.

The problem occurs within any of the Calendar, Timeline and Videos List. The XML Parsing error pops up when opening any of the mentioned windows while the Reference Error only pops up in the Timeline window.

I also noticed that the dropdown "flashed" when clicked (it opens but quickly closes).

[image]

DL44227, 2017-06-29:
I have tried nginx based reverse proxy. Still the same issue, mixed content. So I assume it is not an apache2 related issue.

moeiscool, 2017-07-09:
Please try the latest master

DL44227, 2017-07-10:
No change in the latest dev. Still the same issue.

DL44227, 2017-07-18:
Probably I found a hotfix for this, please see https://github.com/moeiscool/Shinobi/pull/236

closed 2017-07-28 by moeiscool