moeiscool/Shinobi #211: Translation of UI

strixaluco, 2017-06-29:
Hi. Is there any plans to make Shinobi's UI translatable?

moeiscool, 2017-06-29:
like for other languages? you can use google translate. It works pretty well. I may perhaps build a engine to do so but I won't be putting many languages myself and maybe not for some time.

Also please direct your questions in the community chat or website live chat. The github issues is for bugs with the software :) Thanks!

closed 2017-06-29 by moeiscool

strixaluco, 2017-06-29:
Yes, for other languages. 
I meant to use some of the platforms available (e.g. Weblate) so that community members can contribute translations to own language. Didn't mean you have to add them by your own.
Apologies for misusing the tracker.