moeiscool/Shinobi #225: Fullscreen vs thumbnail not fully correct scaled

leeyuentuen, 2017-07-08:
in thumbnail it show the size correctly
[image]

But on full screen it miss a part on the top and buttom?
[image]

moeiscool, 2017-07-08:
Well that is strange. Most likely just an issue with how i full screen Base64 streams. Try another stream type like HLS or JPEG.

Can you please press Export on your monitor settings and post the json here. Please remove any usernames or passwords.

leeyuentuen, 2017-07-09:
this is the export

> {"mode":"record","mid":"IPCAM07","name":"IPCAM07","type":"h264","protocol":"rtsp","host":"192.168.1.226","port":"554","path":"/user=admin_password=xxx_channel=0_stream=0.sdp?real_stream","ext":"mp4","fps":"1","width":"640","height":"480","details":"{\"fatal_max\":\"\",\"notes\":\"\",\"rtsp_transport\":\"tcp\",\"muser\":\"\",\"mpass\":\"\",\"port_force\":\"0\",\"sfps\":\"1\",\"aduration\":\"\",\"probesize\":\"\",\"stream_type\":\"b64\",\"stream_mjpeg_clients\":\"\",\"stream_vcodec\":\"libx264\",\"stream_acodec\":\"no\",\"hls_time\":\"2\",\"preset_stream\":\"ultrafast\",\"hls_list_size\":\"3\",\"signal_check\":\"10\",\"signal_check_log\":\"0\",\"stream_quality\":\"1\",\"stream_fps\":\"5\",\"stream_scale_x\":\"1280\",\"stream_scale_y\":\"720\",\"svf\":\"\",\"stream_timestamp\":null,\"stream_timestamp_font\":\"\",\"stream_timestamp_font_size\":\"\",\"stream_timestamp_color\":\"\",\"stream_timestamp_box_color\":\"\",\"stream_timestamp_x\":\"\",\"stream_timestamp_y\":\"\",\"stream_watermark\":null,\"stream_watermark_location\":\"\",\"stream_watermark_position\":null,\"snap\":\"1\",\"snap_fps\":\"\",\"snap_scale_x\":\"\",\"snap_scale_y\":\"\",\"snap_vf\":\"\",\"vcodec\":\"copy\",\"crf\":\"1\",\"preset_record\":\"\",\"acodec\":\"no\",\"dqf\":\"0\",\"cutoff\":\"30\",\"vf\":\"\",\"timestamp\":\"0\",\"timestamp_font\":\"\",\"timestamp_font_size\":\"10\",\"timestamp_color\":\"white\",\"timestamp_box_color\":\"0x00000000@1\",\"timestamp_x\":\"(w-tw)/2\",\"timestamp_y\":\"0\",\"watermark\":null,\"watermark_location\":\"\",\"watermark_position\":null,\"cust_input\":\"\",\"cust_snap\":\"\",\"cust_detect\":\"\",\"cust_stream\":\"\",\"cust_stream_server\":\"\",\"cust_record\":\"\",\"custom_output\":\"\",\"detector\":\"0\",\"detector_webhook\":null,\"detector_webhook_url\":\"\",\"detector_command_enable\":null,\"detector_command\":\"\",\"detector_command_timeout\":\"\",\"detector_lock_timeout\":\"\",\"detector_save\":null,\"detector_mail\":\"0\",\"detector_mail_timeout\":\"\",\"detector_record_method\":\"del\",\"detector_trigger\":null,\"detector_timeout\":\"\",\"watchdog_reset\":null,\"detector_delete_motionless_videos\":null,\"detector_send_frames\":\"1\",\"detector_fps\":\"\",\"detector_scale_x\":\"640\",\"detector_scale_y\":\"480\",\"detector_use_motion\":null,\"detector_use_detect_object\":null,\"detector_frame\":\"1\",\"detector_sensitivity\":\"\",\"cords\":\"\",\"detector_lisence_plate\":null,\"detector_lisence_plate_country\":null,\"detector_notrigger\":null,\"detector_notrigger_mail\":null,\"detector_notrigger_timeout\":\"\",\"control\":\"0\",\"control_base_url\":\"\",\"control_stop\":\"0\",\"control_url_stop_timeout\":\"\",\"control_url_center\":\"\",\"control_url_left\":\"\",\"control_url_left_stop\":\"\",\"control_url_right\":\"\",\"control_url_right_stop\":\"\",\"control_url_up\":\"\",\"control_url_up_stop\":\"\",\"control_url_down\":\"\",\"control_url_down_stop\":\"\",\"control_url_enable_nv\":\"\",\"control_url_disable_nv\":\"\",\"control_url_zoom_out\":\"\",\"control_url_zoom_out_stop\":\"\",\"control_url_zoom_in\":\"\",\"control_url_zoom_in_stop\":\"\",\"groups\":\"[]\",\"loglevel\":\"warning\",\"sqllog\":\"0\",\"detector_cascades\":\"\"}","shto":"[]","shfr":"[]"}

leeyuentuen, 2017-07-09:
i also got a lot if this error:
could it be the camera issue or ffmpeg encoding issue?

> 2Df5hBE : IPCAM07
FFMPEG STDERR
in a few seconds
Invalid UE golomb code
2Df5hBE : IPCAM07
FFMPEG STDERR
in a few seconds
Last message repeated 29 times [rtsp @ 0xa56760] Too short data for FU-A H264 RTP packet

moeiscool, 2017-07-09:
No its not an ffmpeg issue. It is the base64 stream, the way I'm doing it. It is actually just a canvas element that is being drawn to. If you have ever worked with canvas you'll know how annoying it is to draw in one of those things especially when you want to resize it. Please try using HLS or JPEG as your stream type for now if you need to the full screen.

moeiscool, 2017-07-22:
sorry its been so long, can you try the latest dev commit? You can see the changes here https://github.com/moeiscool/Shinobi/pull/241

moeiscool, 2017-07-27:
im no longer seeing this error.

closed 2017-07-27 by moeiscool