moeiscool/Shinobi #230: NPM install not work from INSTALL/ubuntu.sh

12nick12, 2017-07-12:
Should "apt install curl" and "curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -" be in the script. I had to manually do this to get it to install on Debian 9. This is because NPM isn't in the default repo for Debian.

moeiscool, 2017-07-12:
thanks :smiley:  i just updated the install methods to include these.
http://shinobi.video/docs/start#content-the-ninja-way
http://shinobi.video/docs/start#content-ubuntu-debian--the-easier-way
http://shinobi.video/docs/start#content-centos-fedora-rhel--the-easier-way
http://shinobi.video/docs/start#setup_ubuntu_hard

I wasn't aware it wasn't packaged. There are people saying they have it running on Debian 9 pretty much every week :sweat_smile:

closed 2017-07-12 by moeiscool

12nick12, 2017-07-12:
> thanks 😃 i just updated the install methods to include these.

Awesome. I might have been doing something wrong, but it wasn't working for
me for some reason. Thank you for the quick response :-) have a great day.