moeiscool/Shinobi #231: Auto update DB flag on end time of videos causes unwanted behaviour

DL44227, 2017-07-13:
Clicking on a video in power viewer causes an unwanted update of video end time in the db. It seems to be caused by the SQL definition of this row in the database:

The "end" row in "Videos" table has the attribute "on update CURRENT_TIMESTAMP" set. In consequence each attempt to update a particular video row updates the end time value as well.

After removing this attribute via:

ALTER TABLE `Videos` CHANGE `end` `end` TIMESTAMP NULL DEFAULT CURRENT_TIMESTAMP;

this issue disappeared.

moeiscool, 2017-07-13:
Thank you, I have now committed this to master #232

moeiscool, 2017-07-13:
I should mention anyone who currently has Shinobi installed should run this SQL file `sql/update-13-7-2017.sql`. This will apply Damian's fix.

closed 2017-07-13 by moeiscool