moeiscool/Shinobi #239: possible to have a guest or no password url for viewing

ytjohn, 2017-07-20:
I'd like to put some camera streams into a monitor group, and just provide a url so "anyone with the link" can view it. Is this possible?

DL44227, 2017-07-20:
Yes. Create a sub user with view privileges, login as this user to the dashboard, go to API menu, create a key. Use this key in one of the URLs on http://shinobi.video/docs/api

moeiscool, 2017-07-20:
DL44227 is suggesting that once you have gotten an API key use it to get the list of monitors.

http://shinobi.video/docs/api#content-getting-monitors

Then once you have your list you can get streams based on the returned monitor IDs.

http://shinobi.video/docs/api#content-get-streams

Also please know that the Issues section is not for general questions. It is only used to make note of bugs with the software. Please forward general questions to discord http://discordapp.com/invite/mdhmvuH/

closed 2017-07-20 by moeiscool