moeiscool/Shinobi #245: Doubling file extension (ENOENT in camera.js)

DL44227, 2017-07-24:
I found the following error message in the camera.js log. This message is logged for every cam in regular intervals.

`2017-07-24T12:45:03+02:00 { Error: ENOENT: no such file or directory, stat '/opt/shinobi-dev-20170720/videos/xxx/yyy/2017-07-24T12-39-59.mp4.mp4'
  errno: -2,
  code: 'ENOENT',
  syscall: 'stat',
  path: '/opt/shinobi-dev-20170720/videos/xxx/yyy/2017-07-24T12-39-59.mp4.mp4' }  
`

It seems, that for some reason the file name extension is doubled (.mp4.mp4). Unfortunately this log output does not reference the line where it has occured.

moeiscool, 2017-07-24:
Well that is very bad

moeiscool, 2017-07-24:
i haven't been able to reproduce this, can i get your monitor settings for the camera that this is happening to? or is it happening to all of them?

closed 2017-08-08 by moeiscool