moeiscool/Shinobi #251: License change

mattdurant, 2017-07-28:
I'll preface this by saying that I am not a lawyer, so perhaps my understanding is incorrect, but given the fact that CC-NC is not compatible with GPL, shouldn't the project have been forked to make this drastic of a change?

My understanding is that as the rights holder for code I contribute, in order to make a licensing change you would need to get the permission of all other rights holders to make that change?

moeiscool, 2017-07-28:
if anyone would like to dispute my decision i guess i can always delete this repository and start again?

closed 2017-07-28 by moeiscool

moeiscool, 2017-07-28:
matt and I have discussed the repercussions of my decisions and it will be going back to GPLv3