moeiscool/Shinobi #254: No recognize of hanging ffmpeg process

DL44227, 2017-07-31:
I had a hanging stream today. Other than in cases of other users, in my case the ffmpeg process was still there, but did not do anything: 
 
- It did not send any frames to the detector (I know this because I write a log for each detection)
- It could not be seen in jpeg mode (where the last jpeg frame was displayed)
- Also no video in live mode (just amber colored area was displayed)

This hanging ffmpeg prcoess was started on July 30th and was working for some time (videos were recorded). But after a while probably some network issues occured to the wifi connection and the ffmpeg process stuck. 

A have already the retry setting set to 0. The problem is, that shinobi does not recognize in such situation, that the ffmpeg is hanging.

Suggestion: Maybe restarting the ffmpeg process (configurable time e.g. every n minutes) would help improve this.

moeiscool, 2017-07-31:
I'm actually going to do this in 3 ways, I'm going to watch the stream and if it stops then restart ffmpeg.

Base64 : will reset a timer every time a frame is sent, if that time meets end it will restart
MJPEG : same as base64, slightly different maybe
HLS : reset timer on stream folder change
JPEG : same as HLS

moeiscool, 2017-07-31:
try the latest dev commit, see if it works for you. so far when a stream stops for it just reboots.

https://github.com/moeiscool/Shinobi/commit/5bfc30ad6cffbe8a300c1890c40b00bffb6bf922

moeiscool, 2017-08-02:
going to close this issue unless it comes up again, seems to be ok for me.

closed 2017-08-02 by moeiscool