moeiscool/Shinobi #261: Is the new license setup for Shinobi/Pro correct?

Cozza38, 2017-08-04:
I'll preface this by saying that I am not a lawyer, so clarify if my understanding is incorrect.

I noted today that "Shinobi Pro" is now a thing and took a quick read through the [blog post](http://shinobi.video/articles/2017-08-03-shinobi-ce-and-shinobi-pro). Shinobi Pro has been forked from Shinobi (Now Shinobi CE) and licence changed from GPL to CC BY-NC-SA 4.0.

I'd like to ensure that what has been done is in the best interests of the community.

1. It has been stated in the blog post that Shinobi Pro cannot be used without a "license key" and if the source code is modified to remove the need of a licence key then "things will get legal".
My understanding of [CC BY-NC-SA 4.0] is that anyone may "copy and redistribute the material in any medium or format & remix, transform, and build upon the material". This states that a user can freely modify the source code to remove the need for a license key freely and without recourse. As long as the licence terms are met (Appropriate credits, may not sell, same licence). Source: https://creativecommons.org/licenses/by-nc-sa/4.0/ | https://creativecommons.org/licenses/by-nc-sa/4.0/legalcode

2. My second point is a little hazy but by forking Shinobi directly into Shinobi Pro and changing its licence from GPL that all contributors to the original code must accept the new license or their contributions must be removed. There are a total of 17 contributors on the original code base, many are extremely small changes and their contributions may not even exist anymore but I believe it is a base you need to cover.

3. Is only requiring pull requests onto the Shinobi Pro repository a good idea? You're essentially stating that you are happy for people to commit code to you but only you will profit from it. It is a very large detractor from what makes open source great and will only put more pressure on yourself to create the software as people will be much more hesitant to help. This will give people more incentive to hard-fork Shinobi Pro and make it properly open source again which is allowed by CC BY-NC-SA 4.0.

Lastly, Creative Commons themselves even recommend NOT to use it for Software: https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software

I repeat again as I stated in Discord before you deleted my messages, you should get in touch with a lawyer to find the right licence to meet your goals.

mattdurant, 2017-08-04:
We've covered all of this already, and it's getting ridiculous now.  I regret trying to get involved in this project at all.  Moe, you need to get legal advice on this, all of the changes you are trying to make are against the terms of the license you have been operating under.  I'm sure this is going to be closed without discussion, but at least it'll be here in the record.

closed 2017-08-04 by moeiscool

Cozza38, 2017-08-04:
This isn't a "personal opinion or question".

This is an "issue" related to the license of the software. It is as important as the code itself is.

moeiscool, 2017-08-04:
Choosing to ignore what I say is not acceptable. I am sorry.